makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #285: Problems with Z homing and Z offset

Saigamle, 2021-10-16:
I've been trying to fix it for days, but definetily i cant. I bought a Robin Nano V2 with TS35 and 3dtouch and after lot of forums, youtube videos and google, the machine keep doing the same error.

When i use Z home, or Z offset, its supposed to move the probe to the center of the bed, and use the 3dtouch probe to reach the Z=0, but, in fact, what the machine does, its to move the central carriage to the opposite corner of X=0, Y=0.

Its a COREXY machine

[Configuration.txt](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7356597/Configuration.txt)
[Configuration_adv.txt](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7356598/Configuration_adv.txt)
.

FernandoLopez12, 2022-02-19:
Hi, i'm having the same problemas, when u want to homing z, x and y start going to the Max position, have u solved this?

solawc, 2022-05-16:
Please update the new version to confirm whether it has been repaired

closed 2022-05-16 by solawc