makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #286: Getting MKS Assets doesn't work

gensoloone, 2021-10-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

It is not possible to get a directory with icons and fonts at the beginning of the compilation of marlin. When executing the script download_mks_assets.py an archive is downloaded in which there is no Firmware directory with icons and fonts.

Не удается получить каталог с иконками и шрифтами при начале компиляции marlin. При выполнении скрипта download_mks_assets.py скачивается архив в котором отсутствует каталог Firmware c иконками и шрифтами.

### Bug Timeline

18.10.2021

### Expected behavior

I expect that the problem will be fixed

Я ожидаю, что проблема будет устранена

### Version of Marlin Firmware

Mks-Robin-Nano-Marlin2.0-Firmware

steve007iii, 2021-10-20:
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/278#issuecomment-947981840

gensoloone, 2021-10-21:
Above I wrote how to temporarily solve the problem with missing images and fonts.

I just downloaded the source code again. Made changes for Robin Nano v3.

MKS Robin Nano V3.x build and update firmware
Build config:
platformio.ini:

default_envs = mks_robin_nano_v3_usb_flash_drive

Configuation.h:
#define SERIAL_PORT -1
#define MKS_TS35_V2_0
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
#define TFT_LVGL_UI
#define TOUCH_SCREEN

Configuation_adv.h:
After 2021.6.7, you can use the multi-volume function.
Use the TF card and USB disk together:
#define USB_FLASH_DRIVE_SUPPORT
Only use TF card:
// #define USB_FLASH_DRIVE_SUPPORT

The problem itself is:
[image]
The archive with the code is downloaded. But there is no directory in it from which the assets folder is formed.

[image]

josepatron, 2024-03-15:
I had an issue with Nano Robin V3.1 where the USB flash drive image "icon" did not appear after many compilations and firmware uploads. The speed of uploading firmware I had noticed was very fast deleting and writing ASSETS but suddenly in a last firmware upload it happened really slow, I could read every item erased or copied and when it finished I checked and violá, the USB flash drive icon was there. I have to make a couple changes but I am afraid maybe I loose the icon again.

josepatron, 2024-03-15:
I forget to say that in the last firmware upload the only thing different was that I had an USB flash drive connected to the board. Really rare.