makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #294: [FR] TMC2209 or TMC2209_STANDALONE for configure MKS Eagle Marlin

kroxiksut, 2021-11-05:
### Is your feature request related to a problem? Please describe.

I'm bought MKS Eagle motherboard and me need update firmware.
I'm read instructions for my motherboard, but no says about driver type.
Me need type TMC2209 or TMC2209_STANDALONE for
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
???

### Are you looking for hardware support?

MKS Eagle

### Describe the feature you want

Whats driver type need write for MKS Eagle on 864-866 string in Configuration.h?

phcay, 2021-11-07:
The "TMC2209_STANDALONE" mode is used when there is no UART communication with the TMC2209 driver (eg: MKS Robin nano 1.2). The MKS EAGLE, like the nano V2 and V3, communicates in UART, so you have to use the "TMC2209" mode, so you can configure the current for the motor in software, STEALTHCHOP, SPREADCYCLE or HYBRID_THRESHOLD mode ...

mks-viva, 2021-12-17:
MKS eagle is TMC2209 Uart mode, so, you need set TMC2209 on marlin

closed 2021-12-17 by mks-viva