makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #295: [BUG] Progress bar doesn't work with lvlg ui

Davide655321, 2021-11-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

the progress bar is blocked on 0%.

### Version of Marlin Firmware

02000902

### Printer model

flying bear ghost

### Electronics

moba = robin nano v1.2 driver = tmc2209 in uart 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

phcay, 2021-11-09:
If you are using a print server to stream GCODE to the printer, can you check that the "#define LCD_SET_PROGRESS_MANUALLY" directive is active in your Configuration_adv.h. This activates the M73 command so that the host can update the progress bar. And also check that the host is using this command and not another that is not managed by Marlin.

phcay, 2021-11-09:
Well, looking in the source code, indeed, the progress bar only works when printing is managed locally on the SD card.

However, using the M73 command, and provided the LCD_SET_PROGRESS_MANUALLY and USE_M73_REMAINING_TIME directives are active, it is possible for the host to send the remaining time, and there the MKS code manages and displays this information instead of the elapsed time. , even if it does not manage the percentage of the M73 command for the progress bar.

M73 syntaxe:
```
M73 P<percent> [R<minutes>]
```

R argument only supported if USE_M73_REMAINING_TIME is active

It should also be noted that the elapsed printing time timer is the one managed by the MKS code and is not that of MARLIN.

phcay, 2021-11-10:
One thing to test, would be to replace the following code in the file "Marlin\src\lcd\extui\mks_ui\draw_printing.cpp":
```
void setProBarRate() {
  int rate;
  volatile long long rate_tmp_r;

  if (!gCfgItems.from_flash_pic) {
    #if ENABLED(SDSUPPORT)
      rate_tmp_r = (long long)card.getIndex() * 100;
    #endif
    rate = rate_tmp_r / gCfgItems.curFilesize;
  }
  else {
    #if ENABLED(SDSUPPORT)
      rate_tmp_r = (long long)card.getIndex();
    #endif
    rate = (rate_tmp_r - (PREVIEW_SIZE + To_pre_view)) * 100 / (gCfgItems.curFilesize - (PREVIEW_SIZE + To_pre_view));
  }
```
By this one:
```
void setProBarRate() {
  int rate;
  volatile long long rate_tmp_r;

  if (!gCfgItems.from_flash_pic) {
    #if HAS_PRINT_PROGRESS
      rate = ui.get_progress_percent();
    #endif
  }
  else {
    #if ENABLED(SDSUPPORT)
      rate_tmp_r = (long long)card.getIndex();
    #endif
    rate = (rate_tmp_r - (PREVIEW_SIZE + To_pre_view)) * 100 / (gCfgItems.curFilesize - (PREVIEW_SIZE + To_pre_view));
  }
```
Like this, it should support updating the progress bar via the M73 command, while keeping a more accurate progress bar of the original MKS code when printing with the SD card when a GCODE with a preview image is present (removal of the size occupied by the image).

Please note, I am not a developer and I do not have the possibility to test this myself right away. I'll do it later.

phcay, 2021-11-11:
Finally, from what I understood from the code, TFT_LVLG_UI does not support printing out of SD card, initialized by the interface or by the WiFi module. It is therefore not possible to see the progress or to control (stop, pause ...) a print streamed by a host. So the code I proposed above is useless.
The problem of "Davide655321" must be different from what I assumed. More detail is needed on his problem.