makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #298: [BUG] (bug summary)

albertoraki, 2021-11-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Cuando cambio parámetros de cama o cualquier otro dato y lo compilo con visual studio code, lo abro con platformIO,  y lo compilo con Auto build Marlin, me da alguna advertencia, copio el archivo*.bin en la sd y lo cargo, se reinicia la placa constantemente.
Con la versión que descargo"Mks-Robin-Nano-Marlin2.0-Firmware-master" funciona correctamente, solo es cuando vuelvo a compilar. He probado a compilar sin hacer ningún cambio y lo mismo, se reinicia.
Mi placa es Mks robin nano v3 y la tft es Mks ts35 v2.0
Me podríais decir que hago mal? 
No se puede modificar?
gracias

https://user-images.githubusercontent.com/73721008/141839136-42e2c177-d420-410f-a26c-547fe8cdde72.mp4

### Bug Timeline

cuando compilo

### Version of Marlin Firmware

Marlin bugfix-2.0x

### Electronics

mks robin nano v3

### Add-ons

MKS TS 35 V2.0, TMC2209,

### Additional information & file uploads

Uploading video1_Trim.mp4…

https://user-images.githubusercontent.com/73721008/141839474-3a7b9d2a-44a5-4c62-906e-e1fca66bb265.mp4

https://user-images.githubusercontent.com/73721008/141839567-2cef04b9-2b91-4ebc-b450-ba4de5b3e32b.mp4

phcay, 2021-11-15:
I see that your directories date from 10/25/2021. I think you are using a previous commit which was buggy and repeatedly rebooted the Robin Nano. Redownload the last version, which is corrected on this side. If you still have problems, give the files platformio.ini, Configuration.h and Configuration_adv.h.

albertoraki, 2021-11-15:
creo que este es el último firmware, si hay otro pásame el enlace porque no lo encuentro.
Gracias

https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

albertoraki, 2021-11-15:
[Configuration.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7541510/Configuration.zip)