makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #300: The values in configuration.h are not in the printer

Tr1ckle, 2021-11-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,
Please why my values in configuration.h are not uploaded to the printer?
The compilation is going well. The Robin_nano35.bin file is created. I'll put it on the SD card. I will insert it into the printer, the installation will take place. The file on the SD card changes to Robin_nano35.CUR. But the printer still has old values.
[image]
[image]

### Version of Marlin Firmware

Marlin-2.0.92

phcay, 2021-11-26:
This is normal behavior.
DEFAULT_AXIS_STEPS_PER_UNIT, as its name suggests, is the default value, if you do not have the parameters saved in your EEPROM, or you request their reset. If they are present and compatible with the version of MARLIN that you flashed, they are used. Be careful, if you reset the parameters, you may have surprises, if they are different in your MARLIN configuration, like the Z offset, if you use a Probe.
You may have had the case, where after a MARLIN update, these parameters were updated, but it is that the parameters in the EPPROM were no longer compatible with the new version, so they were automatically reset.

solawc, 2021-12-14:
[image]
You need to change this place, after this change, the relevant value will be updated again @Tr1ckle

closed 2022-05-16 by solawc