makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #301: [BUG] Layer shifting during the print

jhonToni, 2021-12-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

During the print the layer is shifted by the machine.

### Bug Timeline

The bug has occurred with the latest version of the "Mks-Robin-Nano-Marlin2.0-Firmware"

### Expected behavior

I expected that the printed object have not shifted layers.

### Actual behavior

During the print the layer is shifted by the machine.

### Version of Marlin Firmware

Marlin 2.0.9.2

### Printer model

An home made machine

### Electronics

The machine have a "MKS Robin Nano V 1.2" as controller, A4988 stepper motor drivers, nema17 motors, BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Marlin_configs.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7650936/Marlin_configs.zip)
[model.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7650942/model.zip)
[image]
[image]

closed 2021-12-04 by jhonToni

madurani, 2021-12-04:
Duplicate with https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/293

Alex-TV, 2022-01-08:
I have this problem too. mks robin nano v 1.2