makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #302: [BUG] Eeprom write error

telemac73, 2021-12-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After i make a m500 i gote Recv: echo:Error writing to EEPROM! and all data are erased but that dont append all the time.
i dont have this problem with an old marlin 020006 so i think is not a hardware problem.
Thanks

### Version of Marlin Firmware

02000902

### Printer model

Sapphire plus

### Add-ons

Bltouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

solawc, 2021-12-14:
I don't understand what you mean. Did you send M500 through the host computer to execute it?

telemac73, 2021-12-14:
i try m500 with Octoprint and with printer lcd both have the same issue