makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #305: Can LVGL UI show end time?

NNGB-kor, 2021-12-30:
### Is your feature request related to a problem? Please describe.

I am using LVGL UI.

When the output is finished, only the "done" button is displayed along with the guide text.

Couldn't find any function to display end time in LVGL UI.

How can I see the end time?

### Describe the feature you want

I want to display end time in LVGL UI.

MKS-hosiey, 2022-01-03:
Hello, the firmware we will update later will add it

rick090980, 2022-01-14:
Im surprised you were able to get everything working using LVGL UI. I had SO many problems trying to get everything working correctly. I was about to lose my mind when i saw a post saying they used the Marlin Color UI instead of LVGL, tried it and EVERYTHING started working! Give it a shot.

solawc, 2022-05-16:
Please update the new version to confirm whether it has been repaired

closed 2022-05-16 by solawc