makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #306: Compilation error

themacboy, 2022-01-11:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

While compiling I receive and error:

Marlin\src\HAL\STM32\tft\tft_spi.cpp:255:24: error: 'DMA2_Stream3_IRQn' was not declared in this scope
  255 |   HAL_NVIC_SetPriority(DMA2_Stream3_IRQn, 5, 0);

Compilation data:
Board: mks_robin_e3p
Screen: TS35

### Bug Timeline

First time

### Expected behavior

compilation success

### Actual behavior

Compiling .pio\build\mks_robin_e3p\src\src\HAL\shared\HAL_MinSerial.cpp.o
Marlin\src\HAL\STM32\tft\tft_spi.cpp: In static member function 'static void TFT_SPI::TransmitDMA_IT(uint32_t, uint16_t*, uint16_t)':
Marlin\src\HAL\STM32\tft\tft_spi.cpp:255:24: error: 'DMA2_Stream3_IRQn' was not declared in this scope
  255 |   HAL_NVIC_SetPriority(DMA2_Stream3_IRQn, 5, 0);
      |                        ^~~~~~~~~~~~~~~~
*** [.pio\build\mks_robin_e3p\src\src\HAL\STM32\tft\tft_spi.cpp.o] Error 1

### Steps to Reproduce

Donwload MKS repo.
Set platformio.io
Set configurations files
Run compilation

### Version of Marlin Firmware

Current MKS repo version

### Printer model

custom

### Electronics

mks_robin_e3p, TS35

### Additional information & file uploads

th
[configs.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7850206/configs.zip)
x

themacboy, 2022-01-11:
Runing compilation in bugfix-2.0.x  told me that my configs files are too old and can't continue. Not sure aboiut what I must change to success

themacboy, 2022-01-11:
ST STM version 12.0

themacboy, 2022-01-11:
[image]

MKS-hosiey, 2022-01-18:
Hi, did you download the latest marlin bugfix firmware? I tested that firmware is compiled normally

themacboy, 2022-01-20:
> 

Yes, of course. My repot y sync with your repo. And that happends after last merge I do.

themacboy, 2022-01-20:
do you have any idea what the problem might be? Any broken library ?

themacboy, 2022-01-20:
[configs.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7905064/configs.zip)

lethargicpillow, 2022-01-20:
Had succesfully compiled your firmware (with slight changes). Marlin 2.0.9.3
https://github.com/MarlinFirmware/Marlin/releases
[configs.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7905718/configs.zip)
Try to build it, if no success, i'll upload comiled firmware to cloud.

themacboy, 2022-01-20:
> Had succesfully compiled your firmware (with slight changes). Marlin 2.0.9.3 https://github.com/MarlinFirmware/Marlin/releases [configs.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/7905718/configs.zip) Try to build it, if no success, i'll upload compiled firmware to cloud.

@mks-viva yes, running las bugfix from marlin repo with my configs it worked.

themacboy, 2022-01-20:
> Had succesfully compiled your firmware (with slight changes). Marlin 2.0.9.3 https://github.com/MarlinFirmware/Marlin/releases [...]

Thx !!!

Icare2000, 2022-01-25:
Hi i have exactely the same issue,

closed 2022-05-23 by solawc