makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #313: [BUG] (bug summary)

DrSirGe, 2022-02-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Firmware/Robin_nano_v3.bin - on screen all ok!
But if I compiled with default Configuration.h from repository - I have this:
[image]

### Version of Marlin Firmware

default

### Printer model

Anycubic Kossel Linear Plus

### Your Slicer

Cura

jbk9229, 2022-02-28:
I have the same problem.
If you update the firmware to the latest Marilyn version, the screen is not broken, but USB recognition and WIFI do not work.
My guess is that there is a problem with the LCD part adjusted by the manufacturer.

Phisik, 2022-04-06:
Can confirm the problem. I have bought MKS Eagle v1.0 + TS35 v2.0 from Makerbase Global Store (https://aliexpress.ru/store/911610282) (https://aliexpress.ru/item/1005003355188332.html) on 2022-02-21.  Filled rectangles are visible in arbitrary places. 

Some artificial lines and rectangles are also visible in TFT_COLOR_UI mode. But much less.

closed 2022-05-23 by solawc