makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #315: 8.3 filename format does not supported trough wifi upload

pagabee, 2022-03-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If you copy a gcode file trough wifi and if its name is 5 or more characters long the copied filename on SD card will be xxxxx~1.gco.
This is not a 8.3 filename format.

The checking of the lengt of the filename happens in the wifi_module.ccp
[_static bool longName2DosName(const char *longName, char *dosName)_](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/105de25405ff6cc0baca2006c3e158470df4248a/Marlin/src/lcd/extui/mks_ui/wifi_module.cpp#L144)

    if (i >= 5) {
      strcat_P(dosName, PSTR("~1.GCO"));
      return dosName[0] != '\0';
    }

What is the reason for this?

### Bug Timeline

Old bug

### Expected behavior

8.3 filename support

### Actual behavior

qwer.gco --> qwer.gco

qwert.gco --> qwert~1.gco

qwertz.gco --> qwert~1.gco

qwertzu.gco --> qwert~1.gco

qwertzui.gco --> qwert~1.gco

qwertzuio.gco --> qwert~1.gco

### Steps to Reproduce

Please upload a file trough wifi to the SD card.
I'm currently using ESP3D 3.0 and tested it with the MksWifi.bin firmware.

### Version of Marlin Firmware

02000902

### Electronics

MKS Robin Nano V3

### Add-ons

MKS TS35 Display + MKS WIFI modul

### Additional information & file uploads

[image]

pagabee, 2022-03-06:
I modified the code:

    if (i > 8) {
      dosName[6] = '\0';
      dosName[7] = '\0';
      dosName[8] = '\0';
      strcat_P(dosName, PSTR("~1.GCO"));
      return dosName[0] != '\0';
    }

It works, but i don't know if it's correct.

MKS-hosiey, 2022-03-11:
The character limit of the display interface, currently only so many characters can be displayed

pagabee, 2022-03-11:
I copied directly files with long name to the SD card.
The TS35 display shows maximum 12 characters after that comes the ~.

[image]
[image]
[image]
[image]
[image]
[image]

snow79, 2022-05-19:
> I modified the code:
> 
> ```
> if (i > 8) {
>   dosName[6] = '\0';
>   dosName[7] = '\0';
>   dosName[8] = '\0';
>   strcat_P(dosName, PSTR("~1.GCO"));
>   return dosName[0] != '\0';
> }
> ```
> 
> It works, but i don't know if it's correct.

Where you have find this code???

pagabee, 2022-05-20:
> <makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/315/1132012780@
> github.com>

Hello,

I modified the code to this.
Please search for the “~1.GCO” in the wifi_module.ccp file.
There is link in the bug description.

snow79 ***@***.***> (időpont: 2022. máj. 19., Cs, 19:52) ezt
írta:

snow79, 2022-06-01:
> […](#)

yes but... you have solved or not???
i try with your solution, but the problem presist... now it's not 5 char but 8 ....