makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #316: [BUG] custom icons

just-trey, 2022-03-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Custom icons from mks tool do not load. Shows white box with no data. Did the way to build icons change? If so what is the suggested approach?

### Bug Timeline

This version has the issue could be in previous as well

### Expected behavior

Icons to work and not show white boxea

### Actual behavior

White boxes that say no data. Note boot screen customization works fine all other images are broken

### Steps to Reproduce

Load assets into mks tool for robin nano 1.2 Marlin and then download assets. Put SD card in printer and icons do not appear

### Version of Marlin Firmware

2.0.9.2.

### Printer model

Elegoo Neptune 2

### Electronics

Stock ZNP Robin Nano 1.3 (MKS Robin Nano 1.3 clone)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

arpitt21, 2022-03-09:
I had the same issue @just-trey. I have been wracking my brains trying to figure out what the problem is. I tried both just using the 'assets' folder, and splitting it into 'mks_font' and 'mks_pic', but neither worked.

MKS-hosiey, 2022-03-11:
Are you using the lvgl ui of the marin firmware? Its picture is the picture in the assets file, not the picture of mks_pic

just-trey, 2022-03-11:
Yes I am using assets and they seem to LOAD into the machine. The boot screen works fine but no other images show up. I have tried this in both this repo and the Marlin official repo and get the same results.

caedis21, 2022-03-11:
> 

I have the exact same issue. Boot screen works perfectly, but none of the other icons work. I know that Platformio is having issues pulling all the assets from this RP (it says so in the Problems tab on VS Code), but even when I manually copy the assets folder from this RP onto the microSD card and flash it, the icons don't show up.

closed 2022-05-23 by solawc