makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #319: Error trying to compile and enable Advanced Pause feature

n9jcv, 2022-03-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a Creasee CS30.  It has a MKS Robin Nano V2 with the MKS TS35 V2.  The pause functionality is broken.  When you pause, the pause works but the resume causes the extruder motor to stop extruding filament for the rest of the print.

I am trying to work around this, by compiling my own firmware from the 2.0.9.2 branch.  I want to enable Advanced Pause, so that I can then enable and use M600.
When I try to compile enabling Advanced Pause I get this error;
#error "ADVANCED_PAUSE_FEATURE requires a supported LCD 
controller (or EMERGENCY_PARSER)."

### Expected behavior

Why is the MKS TS35 v2 not a supported LCD controller?

Can someone either tell me how to get the MKS LVGL Pause function to work, or how to enable the Advanced Pause feature so that I can ultimately pause and resume my prints as necessary?
Thank You

### Version of Marlin Firmware

2.0 Bugfix and 2.0.9.2

### Printer model

Creasee CS30

### Electronics

stock  MKS Robin Nano V2 and MKS TS 35 v2

### Add-ons

non

### Bed Leveling

No Bed Leveling

### Your Slicer

Prusa Slicer

MKS-hosiey, 2022-03-19:
No need to enable advanced pause with lvgl ui

n9jcv, 2022-03-19:
The pause feature in mks lvgl ui is broken.  After pausing the extruder does not work.  Please tell me how to fix the pause.

solawc, 2022-05-16:
Please update the new version to confirm whether it has been repaired

closed 2022-05-16 by solawc