makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #320: Keep rebooting after any motor movement when compiled with TFT_LVGL_UI

kadirilkimen, 2022-03-22:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Hi,
When I compile the firmware with TFT_LVGL_UI, it always rebooting after any motor movement. This happens even without any change on the firmware. Also LCD visuals are broken.

When I compile same configuration with TFT_COLOR_UI, there is no problem. everything works fine.

Source : Mks-Robin-Nano-Marlin2.0-Firmware with or without any configuration changes.
Board: Robin nano v3

### Bug Timeline

I just started using the firmware and it happened

### Expected behavior

Not to reboot after every motor movement

### Actual behavior

When I try to move any motor, it reboots.

### Steps to Reproduce

Mks-Robin-Nano-Marlin2.0-Firmware
Just compile it for Mks robin nano v3

### Version of Marlin Firmware

2.0.9.2

### Printer model

Custom

### Electronics

Robin nano v3

kadirilkimen, 2022-03-25:
I just compiled main Marlin source code and did all recommended nano v3 configurations and it worked very well. It seems the current mks marlin 2.0 repository has a problem.

awakenedlast, 2022-03-27:
I have this same problem but its random. I can sometimes get a few moves in before it reboot. I'll try your tip. I'm very frustrated with this board at the moment and the lack of resources from makerbase...

solawc, 2022-05-16:
already fixed

closed 2022-05-16 by solawc