makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #325: White screen - FW: MKS update on 11. April

pagabee, 2022-04-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The FW version updated on April 11. by MKS does not work on my board. There is only white screen after switching on the printer.

### Bug Timeline

New bug

### Actual behavior

White screen.

### Version of Marlin Firmware

2.0.92

### Printer model

Custom

### Electronics

MKS Robin Nano V3

### Add-ons

MKS TS35 Display + MKS WIFI modul

MKS-hosiey, 2022-04-13:
Did you update the firmware correctly? Has the picture been updated? Is it normal to update to the old firmware now?

Tenteni, 2022-04-16:
I have the same white screen problem, after flashing the April 11th Solawc https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware. The problem desappears when flashing back the previous firmware version. 
My printer-base is an old Hictop, totally renewed  with a MKS-Robin_Nano_V3 board, TS35 V2.0 Tft_lvgl_ui, 3Dtouch and MKS PWC 3.0, as well as a Matrix extruder.

nguyendinhanh276, 2022-04-16:
I have same problem with MKS Robin Nano v1.2 and MKS Robin TFT35 v1.0 display. After compile, I copy "assets" folder and "Robin_nano35.bin" to flash but the update interface no show, just "No data" icon :( somebody give me a hand..... Please!!!

MKS-hosiey, 2022-04-20:
> I have tried yesterday's solawc bugfix but the white screen problem persits. Compilation runs smoothly with only one warning:
>  "Compiling .pio/build/mks_robin_nano_v3_usb_flash_drive/lib67f/lvgl/lv_core/lv_obj.c.o
>  .pio/libdeps/mks_robin_nano_v3_usb_flash_drive/lvgl/src/lv_core/lv_obj.c: In function 'lv_deinit':
>  .pio/libdeps/mks_robin_nano_v3_usb_flash_drive/lvgl/src/lv_core/lv_obj.c:130:5: warning: implicit declaration of function 'lv_log_register_print_cb' [-Wimplicit-function-declaration]
>  130 |     lv_log_register_print_cb(NULL);
>  |     ^~~~~~~~~~~~~~~~~~~~~~~~"

can you give the bin file? we will test it

Tenteni, 2022-04-22:
Congratulations! I have just tried solawc's bugfix dated 11 hours ago and bingo!!: all problems are gone. 
The assets have loaded fine and two old minor bugs have also been elimininated: one was a difficulty to read  files from the usb flash drive and the other was a problem with the fan pwc regulation. Now everything works fine: Thank you very much indeed!!!

pagabee, 2022-05-09:
The bugfix on 22 April solved the problem.

closed 2022-05-09 by pagabee