makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #332: not compiling with MKS_Robin_nano35 for MKS Robin Nano v1.2

metrolan, 2023-01-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I've tried to compile making the changes stated for platformio.ini, Configuration.h and Configuration_adv.h, and I get error: 
`Error: Build environment 'mks_robin_nano35' is incompatible with BOARD_MKS_ROBIN. Use one of these: mks_robin, mks_robin_maple`
Using vscode (Linux) 1.74.2, Platformio core 6.1.5 Home 3.4.3 and Configuration.h, Configuration_adv.h attached.
In PlatformIO selected mks_robin_nano35. Tried mks_robin_nano and mks_robin_nano_maple giving lots of errors.

Is needed any change not mentioned in the readme.md?

### Expected behavior

I expected to compile without  errors.

### Actual behavior

It didn't compile

### Steps to Reproduce

Use the files mentioned 

### Version of Marlin Firmware

2.0

### Printer model

Two Trees Bluer Plus

### Electronics

MKS Robin Nano v1.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

[platformio.ini.pdf](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/10363800/platformio.ini.pdf)
[Configuration_adv.h.pdf](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/10363801/Configuration_adv.h.pdf)
[Configuration.h.pdf](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/10363802/Configuration.h.pdf)

giordy72, 2023-01-07:
This works for me with Marln 2.1.2

- Config.h :
    #define MOTHERBOARD BOARD_MKS_ROBIN_NANO

- platformio.ini :
  default_envs = mks_robin_nano_v1v2

I hope this help

metrolan, 2023-01-07:
Thank you.

What did you select to build in PlatformIO?
Using build from mks_robin_nano35 I get this:
`Error: Unknown environment names 'mks_robin_nano_v1v2'`